305engineering/Inkscape #1: Laser not turning off

ripmyundies, 2015-04-07:
Hi, great plugin - really the best out there.

I'm finding the laser doesn't turn off at the end of the 'print'.  I'm assuming it's just not sending the turn off command

oni305, 2015-04-25:
[image]
Strange i've try with this image with B/W and it turn off
"M03; Laser ON
G01 X0.0 Y2.3 F200
M05; Laser OFF
G00 X0 Y0; home"

even with grayscale
"M03  S160 ; Laser ON
G01 X0.0 Y2.3 F200
M05; Laser OFF
G00 X0 Y0; home"

oni305, 2015-04-25:
by the way  I'have reuploaded the file, try it now.

ripmyundies, 2015-05-01:
Thank you sir.  The GCode looks correct, for some reason at the end  of the 'print' the laser remains on.  I'll investigate further and try some manual updates to the gcode to identify the cause

closed 2015-05-05 by oni305