305engineering/Inkscape #2: Extension does not work

meatom, 2015-04-24:
The extension does not generate the gcode or the preview, it just stays "processing..." Any tips?

Thank you, Bernardo.

oni305, 2015-04-24:
have you selected a valid directory for the ouput?
Size of the image? pixell/mm selected? -->if the image is large it can take a looong time to process

meatom, 2015-04-24:
Hello,
Thanks for your reply.
The image is quite small, it's 20x20mm, and I'm using the c:\  for destination.
Pixel/mm is 2/mm.
I tried just writing a letter(like, letter B) in Inksacape but it does not work as well.
Do you think that a 20x20mm image can take an hour to process?

Regards.

meatom, 2015-04-25:
Which version of Inkscape are you using?

oni305, 2015-04-25:
This dimension is fine.
I use inkscape v 0.485 r10040  on  win7

try to create a directory on the desktop and use that

meatom, 2015-04-25:
Hello again,
I had success with an old version of your extension, I got it from this web site: http://www.openbuilds.com/builds/laser-engraver-control-board-inkscape-extension.515/.
I'm using the v6 and it works fine. But I really don't know why the version from Github does not work :-(.
Anyway if you have some idea...
I really like your project, it fits perfectly to my mini laser engraver made with DVDRom laser in my 3D printer!

Thankyou!

oni305, 2015-04-25:
Thanks for the info!
I'have reuploaded the file and now it should work!!
( I think github "cut" the last line in the file png.py )

meatom, 2015-04-25:
Hello,
Now it's working good! Thank you!

closed 2015-04-26 by oni305