305engineering/Inkscape #5: I can't use this extension to create file.

saveyourtime1990, 2015-09-22:
[image]
I tried to install on 3 computer but It didn't work?
Please help me.

oni305, 2015-09-22:
From this image i can't undersand, can you send me the complete error?

Are you sure to own the right privileges for the folder?
Which OS do you use?
Which Inkscape version do you use?

saveyourtime1990, 2015-09-23:
I use Windows 7 32bit

This is error code:
Traceback (most recent call last):

  File "raster2laser_gcode.py", line 564, in <module>

    _main()

  File "raster2laser_gcode.py", line 559, in _main

    e.affect()

  File "C:\Program Files\Inkscape\share\extensions\inkex.py", line 268, in affect

    self.effect()

  File "raster2laser_gcode.py", line 159, in effect

    self.PNGtoGcode(pos_file_png_exported,pos_file_png_BW,pos_file_gcode)

  File "raster2laser_gcode.py", line 203, in PNGtoGcode

    reader = png.Reader(pos_file_png_exported)#File PNG generato

  File "C:\Program Files\Inkscape\share\extensions\png.py", line 1376, in __init__

    self.file = open(kw["filename"], "rb")

IOError: [Errno 2] No such file or directory: 'd:\\test_0001.png'

herculine, 2015-09-23:
Hi, thanks for the plug in - sure it will be fantastic once I manage to export a file!

I've just tried to create a file and having the same error as mentioned above, running Win7 on 64 bit system, version 0.91 of Inkscape (64bit).

Error message as below:

Traceback (most recent call last):
  File "raster2laser_gcode.py", line 564, in <module>
    _main()
  File "raster2laser_gcode.py", line 559, in _main
    e.affect()
  File "C:\Program Files\Inkscape\share\extensions\inkex.py", line 268, in affect
    self.effect()
  File "raster2laser_gcode.py", line 159, in effect
    self.PNGtoGcode(pos_file_png_exported,pos_file_png_BW,pos_file_gcode)
  File "raster2laser_gcode.py", line 203, in PNGtoGcode
    reader = png.Reader(pos_file_png_exported)#File PNG generato
  File "C:\Program Files\Inkscape\share\extensions\png.py", line 1376, in __init__
    self.file = open(kw["filename"], "rb")
IOError: [Errno 2] No such file or directory: 'C:\\output\\VJ test.png'

Thanks very much.

oni305, 2015-09-23:
I can't recreate the error, from the last line it seems a problem of permission

Are you sure to own the right privileges for the folder?
Do you execute inkscape as administrator?

saveyourtime1990, 2015-09-24:
Yep, I tried to install it on Win 7, win 8 , Mac os, Ubuntu.
It didn't work with win 7 , win 8. Mac os and ubuntu are oke.
I executed inkscape as administrator but nothing changed.

oni305, 2015-09-24:
strange I use win7 64bit and it works... Try the 32bit version of inkscape

saveyourtime1990, 2015-09-25:
Oke, I will try it again.

saveyourtime1990, 2015-09-25:
I tried to use win7 64bit but it didn't work. I don't know what happen. It's driving me crazy!!!

oni305, 2015-09-25:
have you tried the 32bit version of inkscape?
(portable  https://inkscape.org/en/gallery/item/4460/InkscapePortable_0.91.paf.exe )

saveyourtime1990, 2015-09-25:
I've tried all. But no thing changed :( Why it happen to me?

closed 2015-10-20 by oni305