305engineering/Inkscape #7: strange behaviour on the gcode

wasteinc, 2015-11-06:
Im trying to raster some images on my DIY laser cutter but the gcode interpreter makes this strange lines.

If you look closer you will notice that the faulty lines are actually inverted . ie where the laser is off is on and the other way around. 

So the faulty lines are quite precise, so I thought its the interpreter?

thank you for this very nice software

[image]
[image]

oni305, 2015-11-06:
strange...
try to inspect the gcode with a gcode viewer (es:  https://www.cncedit.com/Basicviewer/ )
or send me the image file & the generated gcode

wasteinc, 2015-11-07:
I also found it strange :) 
the basicviewer through wine hits some error about wrong parameters (is there an alternative in linux? I tried a couple but they are 3d printed oriented they dont seem to understand the M016 M107 laser on/off command)
 
so Im sending you the two files one in paste bin
http://pastebin.com/pr0E36PP

and one here

[image]

Idont need to do anything specific on the bitmap image (like paths or anything similar), do I?

oni305, 2015-11-07:
The Gcode seems correct!!!
I think its a problem of the gcode interpreter (or the parser) of your board.
(on linux you can try EMC2 to inspect the gcode) 

[image]

wasteinc, 2015-11-07:
thanks for pointing out on the right direction. I was trying to inspect the code manually but obviously it didnt work :)

I will continue my experiments and see what goes wrong

wasteinc, 2015-11-12:
Ok after 1,5 days of experimenting I reached the root of the problem

The problematic lines were mostly inverted. That mean where should be black was white and vice versa
furthermore the problematic lines were consistent in each run which was helpfull

Ive read that M106 M107 commands are not always in sync with movements. So I tried to add a G4 P60 command before each M106 M107 and voila the problem disappeared :)

j-tech in his own inkscape gcode interpreter has the option to add the G4 PXXX command but it can easily be added with regular expressions like

replace M106
with G4 P60\nM106

and replace M107
with G4 P60\nM107

I hope it helps 
thanks again

wasteinc, 2015-11-20:
New tests :) 

So the pause before an M106 M107 command can also be 0 or just 5 (miliseconds). This is enough for the commands to arrive in sync

The pause AFTER an M106 (laser on)  command can vary according to your needs. Some times lasers need a little bit of time to start engraving (specially if they are not so strong as mine). So there adding a Pause command  of G4 P100-200 will give the laser some time to start . 

Apply if your engraving edjes are not so well defined.

closed 2015-11-20 by wasteinc