305engineering/Inkscape #9: G0/G1 for smoothie

cojarbi, 2016-01-23:
Hi. I have a smoothie board and trying to get your pluggin working but not getting any results. Smoothie expects only G0/G1 parameters. The S parameter should be inline with the movement. 

Is there anyway i can make those adjustmest?

Thanks

cojarbi, 2016-01-23:
Also is exporting the whole inskcape document not just the image

oni305, 2016-01-24:
mmm... it sems to be supported from smoothieboard : ( http://smoothieware.org/supported-g-codes )
" M3 	Starts the spindle. Only if spindle module is enabled. The S parameter sets the speed in rotations per minute "
i think the poin is to enable the spindle module, but i don't have a smoothieboard so i can't help you.
 
My pluggin create an image and a gcode  from the whole document, if you want to save the document, save it as a normal inkscape file (File->Save As..)

oni305, 2016-01-24:
(btw: don't use white to fill the background but leave it transparent)

cojarbi, 2016-01-24:
M3 is on spindle for cnc. For laser it uses S parameter on a move command G1/G2/G3 laser On, G0 laser. Off. 

Your gcode creates double Gcommands if I put G1/G0 on the fields,  and empty goes to your default M3/M5.

Btw I'm also testing a new solution called LaserWeb. The developer could use your help.  It's pretty nice and very complete. 

https://github.com/openhardwarecoza/LaserWeb

Ariel Yahni

oni305, 2016-01-24:
if you send only the S parameter the laser turn on or you absolutely need the G01?

cojarbi, 2016-01-24:
Here you have a sample gcode from a smothie board connected to laserweb

[GCODE-LaserWeb (6).gcode.txt](https://github.com/305engineering/Inkscape/files/102326/GCODE-LaserWeb.6.gcode.txt)

wasteoinc, 2016-01-24:
Hey I use a reprap board and I use the M106 / M107 command.
I  dont know if this helps, but I work like this

I take the code from oni' s really nice plugin and the just add with regular expressions an M106 to add a pause (which is needed with M106/M107 in order not to create artifacts and then give some time to the laser to start burning)

its an extra step but an easy one and you can do it like replace for every G1 to  G1 S255
I dont think your feeder or your board would mind the double command

closed 2017-02-08 by cojarbi