305engineering/Inkscape #14: PWM min / max values

martindb, 2016-12-12:
It should be great you can specify pwm minimun and maximun values. In lasers with high power (mine is 5.5W) the engraved imaged are to dark.

JBSchueler, 2017-02-17:
I ran into the same issue.
Whould be nice to add a minimum and maximum value for PWM when exporting to grayscale.

JBSchueler, 2017-02-18:
In addition, I just upgraded my GRBL to th latest release 1.1f.
This version has some really nice new configurations:

**Max spindle speed, RPM**
```$30=1000```
Can be used to adjust the laser maximum. If set to 1000 the output of the laser will be 25% (S255)

**Min spindle speed, RPM**
```$31=15```
Can be used to adjust the laser minimum. If set to 10 the output of the laser will be 1%.
S1 to S9 will be overwritten by the GBRL code to S10.
Laser is off for S0.

**Laser mode, boolean**
```$32=1```
This is a really nice new setting. Set this value to 1 for lasers. Normally ($32=0) the movement stops if the spindle speed changes but this is only needed for CNC engervers.
For laser you want the movement continue at the same speed so set this value to 1 ($32=1)!.

More information about the new configurations of GRBL 1.1f can be found on their [new Wiki](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration).

The minimum and maximum values have to be find out by experimenting, but these values are a good start. It depends on the power of the laser and type of material.

Code and firmware can be found on the [new github GRBL location](https://github.com/gnea/grbl)

_With these new configurations this ticket is only valid for GRBL versions prior 1.1._

JBSchueler, 2020-04-15:
I wrote some code what makes it possible to set the minimum and maximum value.
See #27