305engineering/Inkscape #16: Coolant ON/OFF

JBSchueler, 2017-02-18:
It would be nice if coolant functionality can be added like the Laser ON/OFF Commands
E.g. at the beginning of the file turning coolant on and at the end off.

```
Enable Coolant      (v)
Coolant ON Command  [ M08      ]
Coolant OFF Command [ M09      ]
```

xxValiumxx, 2017-02-18:
While options are good, I fail to see the point of this, as you will want the coolant running at all times.

JBSchueler, 2017-02-18:
@xxValiumxx : True!

But I miss the option and have to manually add this to the generated file.
Would be nice if this option can be added.
Maybe just the ON and OFF command. When empty it can be ignored.
```
Coolant ON Command  [ M08      ]
Coolant OFF Command [ M09      ]
```

The fan is controlled by software, it doesn't have to be on when it has finished.