305engineering/Inkscape #18: problems with gray scale

DarkRabbit19, 2017-09-29:
dear oni305,

When i was printing with my laser engraver, the gray scale dit not work wel.
but during my test's I got better resolts with different speeds.
Zo i was wondering if jou can change the code to work with speed, instead of total laser Pouwer.
(ore make it a optoin)

I hope to hear from you soon

yours truly,
A maker

JBSchueler, 2017-09-29:
Hi DarkRabbit19,

I made some changes to the code a few month ago to at VFR.
Please put these 2 files in the some directory besides the original ones.
Once Inkscape started, you will see a new version  in extensions.

[image]

Just select Greyscale FRC and set the maximum and minimum feed rate.
max is white, min is black.
Keep in mind this is an experimental version.

[raster2laser_gcode_new2.zip](https://github.com/305engineering/Inkscape/files/1345136/raster2laser_gcode_new2.zip)

DarkRabbit19, 2017-10-01:
dear oni305,

When i was printing testet your new code but it dit not work but this is wat i notist:

* At firtst the print dit not start, but it was "printing" when i put a G1 infront of the F**** X*,* lines.
* When it dit print, thare was no gray scale only black.
* Than i was experiminting whit 2 pixel/mm but then the laser turns on (dit not move) and after a wile goes off and auto home's.

I hope this is helpful with your testing.

yours truly,
A maker

(the test files)
https://www.dropbox.com/sh/3fnqt4cns11taqy/AABVDf91-hqlEbPOW9-yTMZka?dl=0

JBSchueler, 2017-10-01:
Uhm... you are wright.
I tested at my laser and seems I forgot something.
Keeps me wondering way it was working for me at that time.

Please add the following commands at the top for now.

I will make new version tomorrow...

```
G28; home all axes
M05
G00 X0.0 Y0.0
G01 F1574 S255
M03
```

It seems that `G01` needs an `Fxxx` value at the start, it doesn't matter what value it is.
`G01 Fxxx S255`

The `S255` means maximum laser power.
You can lower this value if you want.

[JBSchueler_Test.nc.txt](https://github.com/305engineering/Inkscape/files/1347570/JBSchueler_Test.nc.txt)

JBSchueler, 2017-10-02:
This version has the correction for the laser.
Laser max value will be set at the beginning.
Make sure that `Engarving speed` and `Engarving speed FRC max` have the same value.

Keep in mind that white should be a high feed rate. In my case it is around `3200` and dark is `288`.
Also the acceleration of `X` and `Y`of your CNC laser should be at least double the maximum feed speed.
I use these settings for my 300mW laser.

```
$110=3200.000
$111=3200.000
$112=3200.000
$120=6400.000
$121=6400.000
$122=6400.000
```

A also tuned the algorithm a bit to make more contrast for black and white.

The setting I used for Raster 2 Laser are these
[image]

This results in a picture like this (during cnc laser)
[image]

This is the new version of the code I used.
[raster2laser_gcode_new2.zip](https://github.com/305engineering/Inkscape/files/1350485/raster2laser_gcode_new2.zip)

Naruto128, 2018-02-07:
hello, thank you for your contribution.
You can send me the code for the repetier software. Thank you.

JBSchueler, 2018-02-07:
Hi @Naruto128, I am not sure what you main by "repetier software".
The code is available in my previous post [[raster2laser_gcode_new2.zip](https://github.com/305engineering/Inkscape/files/1350485/raster2laser_gcode_new2.zip)].

Naruto128, 2018-02-07:
https://forum.repetier.com/discussion/1387/implementing-laser-intensity

eckirchn, 2018-03-19:
I have had zero success with either your published version of the above mentioned version.  I realize that does not add much value, but thought I would comment anyway.

marceloXongonzalez, 2021-01-11:
????

Traceback (most recent call last):
  File "raster2laser_gcode_new2.py", line 37, in <module>
    import png
  File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/png.py", line 1885
    except ValueError, e:
                     ^
SyntaxError: invalid syntax