305engineering/Inkscape #26: Inkscape 1.0 support

magdesign, 2019-09-24:
Any chance to get support for **Inkscape 1.0**:
https://inkscape.org/release/inkscape-1.0beta1/

Currently when using the script, getting following erros:

```
raster2laser_gcode.py:50: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("-d", "--directory",action="store", type="string", dest="directory", default="/home/",help="Directory for files") ####check_dir
raster2laser_gcode.py:51: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("-f", "--filename", action="store", type="string", dest="filename", default="-1.0", help="File name")
raster2laser_gcode.py:52: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("","--add-numeric-suffix-to-filename", action="store", type="inkbool", dest="add_numeric_suffix_to_filename", default=True,help="Add numeric suffix to filename")
raster2laser_gcode.py:53: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("","--bg_color",action="store",type="string",dest="bg_color",default="",help="")
raster2laser_gcode.py:54: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("","--resolution",action="store", type="int", dest="resolution", default="5",help="") #Usare il valore su float(xy)/resolution e un case per i DPI dell export
raster2laser_gcode.py:58: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("","--grayscale_type",action="store", type="int", dest="grayscale_type", default="1",help="")
raster2laser_gcode.py:61: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("","--conversion_type",action="store", type="int", dest="conversion_type", default="1",help="")
raster2laser_gcode.py:64: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("","--BW_threshold",action="store", type="int", dest="BW_threshold", default="128",help="")
raster2laser_gcode.py:65: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("","--grayscale_resolution",action="store", type="int", dest="grayscale_resolution", default="1",help="")
raster2laser_gcode.py:68: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("","--speed_ON",action="store", type="int", dest="speed_ON", default="200",help="")
raster2laser_gcode.py:71: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("","--flip_y",action="store", type="inkbool", dest="flip_y", default=False,help="")
raster2laser_gcode.py:74: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("","--homing",action="store", type="int", dest="homing", default="1",help="")
raster2laser_gcode.py:77: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("","--laseron", action="store", type="string", dest="laseron", default="M03", help="")
raster2laser_gcode.py:78: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("","--laseroff", action="store", type="string", dest="laseroff", default="M05", help="")
raster2laser_gcode.py:82: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("","--preview_only",action="store", type="inkbool", dest="preview_only", default=False,help="")
raster2laser_gcode.py:560: DeprecationWarning: Effect.affect is now `Effect.run()`. The `output` argument has changed.
  e.affect()
raster2laser_gcode.py:94: DeprecationWarning: self.args[-1] is now self.options.input_file
  current_file = self.args[-1]
Traceback (most recent call last):
  File "raster2laser_gcode.py", line 565, in <module>
    _main()
  File "raster2laser_gcode.py", line 560, in _main
    e.affect()
  File "/tmp/.mount_InkscabsLWXb/usr/share/inkscape/extensions/inkex/deprecated.py", line 181, in affect
    return self.run(args=args)
  File "/tmp/.mount_InkscabsLWXb/usr/share/inkscape/extensions/inkex/base.py", line 111, in run
    self.save_raw(self.effect())
  File "raster2laser_gcode.py", line 160, in effect
    self.PNGtoGcode(pos_file_png_exported,pos_file_png_BW,pos_file_gcode)
  File "raster2laser_gcode.py", line 204, in PNGtoGcode
    reader = png.Reader(pos_file_png_exported)#File PNG generato
  File "/home/magdesign/.config/inkscape/extensions/png.py", line 1376, in __init__
    self.file = open(kw["filename"], "rb")
IOError: [Errno 2] No such file or directory: '/home/magdesign/Desktop/ragaz_merci_0001.png'
```

wowotek, 2020-01-01:
bump
```python
Traceback (most recent call last):
  File "tugas_akhir.py", line 50, in <module>
    tugas_akhir.affect()
  File "/snap/inkscape/5874/share/inkscape/extensions/inkex.py", line 289, in affect
    self.effect()
  File "tugas_akhir.py", line 38, in effect
    e.affect()
  File "/snap/inkscape/5874/share/inkscape/extensions/inkex.py", line 289, in affect
    self.effect()
  File "/home/wowotek/snap/inkscape/5874/extensions/raster2laser_gcode.py", line 163, in effect
    self.PNGtoGcode(pos_file_png_exported,pos_file_png_BW,pos_file_gcode)
  File "/home/wowotek/snap/inkscape/5874/extensions/raster2laser_gcode.py", line 207, in PNGtoGcode
    reader = png.Reader(pos_file_png_exported)#File PNG generato
  File "/home/wowotek/snap/inkscape/5874/extensions/png.py", line 1376, in __init__
    self.file = open(kw["filename"], "rb")
IOError: [Errno 2] No such file or directory: '/home/wowotek/exportergrtg.gcode_0001.png'
```

TheZeroBeast, 2020-01-30:
I also would appreciate 1.0 support.

magdesign, 2020-05-06:
Anyone able to update the code ?
Anyone else willing to donate ?

I am in with $20.-- via paypal for a fixed version of this code...

bferrarese, 2020-05-07:
For those that need it to work with inkscape 1.x...

https://github.com/bferrarese/raster2laser_gcode

I have never used it before so I don't know the expected outcome. It looks like it is doing what it is supposed to do. 
I like the extension and the idea behind it and I'll make some changes in the future.

magdesign, 2020-05-07:
@bferrarese 

Thanks, just tested on Ubuntu 18.04 with Inkscape.appimage
Installed in:` ~/.config/inkscape/extensions`

The extension is listed, but greyed out.

After digging a bit, figured out that there is no `inkex.py` on my system.
Might be a bug on Inkscape.appimage side or needs the extension more care?

https://gitlab.com/groups/inkscape/-/issues?scope=all&utf8=%E2%9C%93&state=opened&search=inkex.py

magdesign, 2020-05-07:
Now stuck in a encoding error, do we need encodings?

[image]

bferrarese, 2020-05-07:
It does sounds like there is something wrong with the appimage. I don't use Ubuntu/appimage so I can't tell for sure. I use archlinux and there is a package for inkscape already.
```
pacman -S inkscape
```

bferrarese, 2020-05-07:
Yep...Tried the appimage and got the same error.

magdesign, 2020-05-07:
ok. Thanks for the feedback, think its a known issue:
https://gitlab.com/inkscape/inkscape/-/issues/1306
https://gitlab.com/inkscape/inkscape/-/merge_requests/1954

They are on it. Will check back in a few days if it works as expected on the lasermachine.

magdesign, 2020-05-18:
@bferrarese:

Thanks for looking into this. Installed Inkscape1 now from ppa and tested your modification.
It generates the G-code but there is an issue with the offset. The lasermachine complains that its outside the work area. So I compared the output from the original script on Inkscape 0.92 and your script. The first line of code (after the header) in your script, always writes:
`G0 X0.0 Y0.0`

and also all the other coordinates differ a lot. Attached a .zip with the .svg file, the output from 305enineering original and from your mod:
[cnc_test.zip](https://github.com/305engineering/Inkscape/files/4643574/cnc_test.zip)

(Ignore that I changed G00 to G0, since its a modification my machine needs...)

magdesign, 2020-05-18:
btw. should I open an issue on your github page instead of here?

bferrarese, 2020-05-18:
I knew sooner or later I would have to touch its logic. I'll see what I can do.
Yes please, open a new issue on my repo.

magdesign, 2020-05-18:
ok.
please enable issues then on your fork:

Github => Settings => features/Issues

bferrarese, 2020-05-18:
good point

magdesign, 2020-05-18:
I close this issue since we move on here: https://github.com/bferrarese/raster2laser_gcode/

closed 2020-05-18 by magdesign