305engineering/Inkscape #27: LASER STRENGTH ADJUSTMENT

quotelawrence, 2020-01-20:
I have recently downloaded and used your extension to create a laser g-code program, though I did not see any area to adjust or minimize laser strength so I am writing hoping you might be able to inform me of certain settings that would possibly assist in setting conditions, heat, strength, of the laser, I downloaded another program Lasergrbl through the images are poorly transferred and the Chinese program which came with the machine is still possibly the best at producing images and text but I like inkscape and hope this is not a bad subject

JBSchueler, 2020-01-20:
The Black and White values are hard coded in 'raster2laser_gcode.py' at [line 271](https://github.com/305engineering/Inkscape/blob/d14935f56fe112c1a6c6b56f5d88f24f70e09fc7/raster2laser_gcode.py#L271)

B is White and N is Black value.
It is possible to make these variable available in the GUI.
Or for a quick try you can change those values in the code.
You must restart the program after changing the values.

tuxun, 2020-01-20:
Hi

That's mainly why i would love mashup lasergrbl and one of my favorites
inkscape extension, being both opensource, it would be totally doable :)

JBSchueler, 2020-01-21:
[Inkscape-raster2laser_variablelaser.zip](https://github.com/305engineering/Inkscape/files/4093978/Inkscape-raster2laser_variablelaser.zip)
This is the modified code I wrote a few years back to solve the issue you describe.
Don't mind the additional features I implemented, just ignore them.

quotelawrence, 2020-01-21:
yes I downloaded the variablelaser.zip and ran the program I may need to change some settings but It does have max & min laser settings which help not to over load and module thank you