305engineering/Inkscape #28: GCODE File Outputted as TXT File

Andyboy2, 2020-02-13:
I downloaded this extension and ran it in Inkscape and tried implementing it into my laser engraver, but found out it wouldn't identify it because it outputted as a .txt file instead of a .gcode file. Am I doing something wrong? Or do I need to convert this in a separate step after running the extension?

JBSchueler, 2020-02-13:
Just rename the extension to what you need, it is hardcoded.

See [raster2laser_gcode.py](https://github.com/305engineering/Inkscape/blob/d14935f56fe112c1a6c6b56f5d88f24f70e09fc7/raster2laser_gcode.py#L149) line 149.

If you don't want to change the extension all the time, change `.txt` to `.nc` at line 149.

Andyboy2, 2020-02-22:
Thanks! This helped a ton! I changed it to ".gcode" and everything is working well for me now.