305engineering/Inkscape #32: except ValueError, e:

hariharan382, 2021-06-23:
Traceback (most recent call last):
  File "raster2laser_gcode.py", line 37, in <module>
    import png
  File "C:\Program Files\Inkscape\share\inkscape\extensions\Inkscape-master\png.py", line 1885
    except ValueError, e:
                     ^
SyntaxError: invalid syntax

i am getting this error when trying to use gcode extension,i have no idea how to edit it since it is 2500 line of code

FakeBigBoi, 2021-07-20:
im having this same error

WhamBamSystems, 2021-09-29:
I'm getting the same error with Inkscape 1.1 on a Mac.  Does this plugin work for version 1.1?

BCChad, 2021-12-30:
I'm trying to overcome this as well - in my researching ValueError, it seems that it should have no arguments, simply ValueError:
I have removed the ", e" and made progress in debugging, but encountered further errors elsewhere. (newbie at Python, trying my best...).

rktech-de, 2022-01-10:
the above message, that I have added a commit to the issue 32 is wrong. what I did is implementing the pull request 32

Terak8510, 2023-04-07:
Another DEAD project, Thanks to Python3.

WhamBamSystems, 2023-04-07:
Hello,
Is there something Wham Bam related I can help you with?
thanks

my best,

Peter
founder Wham Bam Systems LLC

290 North Federal Highway
Hollywood, FL, 33020   USA
+1 (954) 391-8921

Wham Bam thanks you ma’am!

www.whambamsystems.com
***@***.***

[whambamhorizshadowsmall.jpg]

Terak8510, 2023-04-07:
[image]

[image]

rktech-de, 2023-04-07:
Maybe this helps, I have get my version running with Inkscape 1.1 (Python3) with this change in png.py
https://github.com/rktech-de/Inkscape/blob/master/png.py

Terak8510, 2023-04-08:
Well i am running inkscape 1.0.2-4 and will stay that way, until debian bullseye
finalizes at oldstable.

[image]

rktech-de, 2023-04-08:
Hi, the png.py file shouldn't have any "DOCTYPE" tag, maybe you have downloaded the github website and not only the Python code of the png.py file? 
best you download the full projekt zip https://github.com/rktech-de/Inkscape/archive/refs/heads/master.zip and copy the png.py file (or try it with my fork)