305engineering/Inkscape #34: Vertical Raster feature needed

BCChad, 2021-11-14:
Thank you for your code, which gives me good results.  My suggestion - the raster lines run horizontally in the x-direction. When a piece requires me to rotate it 90 degrees in the machine, the raster lines now are vertical which is not always a pleasing look.  It would be good to have a feature option to run the raster lines in the y-direction.
Thank you.

rktech-de, 2022-01-08:
Hi, because of other changes I need, I did the GCode generator completely new, but it now is also possible to change also the scan direction (tab: G-Code, config: Scan image lines ...). If you like you could be my first external tester of this new code.
Take a look here: https://github.com/rktech-de/Inkscape

BCChad, 2022-01-09:
Yes, thanks. I am happy to test this.
Did you see my additions of dwell (G4), and scaling the the laser power S?
I find the latter was marginally useful.

Bruce

BCChad, 2022-01-09:
First impression - it loads well into inkscape, and the features are
impressive.  I'll be busy trying these out (enthusiastically!).

So it is for Inkscape .92 or should it also work for 1.1.?    I know I was
having problems with png.py in Inkscape 1.1

Thanks for now, continuing...
Bruce

BCChad, 2022-01-09:
Hello,
Here are my comments thus far, in no particular order
1.  Comment to self - I am unfamiliar with the M62, 63, 65, 66, 67, 68
digital controls.  This may give me additional capability I don't
understand right now - my system is MACH3 with a "simple" Genmitsu 3018-MX3
control board (non GRBL).  Please tell me a little about this aspect of
your implementation.
2. Resolution steps - I suggest adding 3,4 pixels per mm (as I did in my
fork) - easily done.. Perhaps also 7 or 8.
3.  FLIP X, FLIP Y - the gcode is flipped but not the preview .PNG  -
consider the advantage (or confusion, if not doing it)
4.   Full Page - with it selected you print everything in the page, as I
expect.  With it unselected, you printed objects outside the page boundary,
even though they were not selected. Is this intentional?
5.  Scan image lines - I like the feature, and the different options.
However- my 40x48mm test image was oriented at 0,0 at the left bottom
corner.  When I selected "Zig zag Y", it started the image at x47 and
finished at x86 - something buggy here.  Some of the other options were
also offset to the right.
6.  My system presently does not utilize M62...etc.   I used the "Power
Level Change Code" field and modified it to use the M03 S command.  This
works, however there is no POWER OFF (M5) command found in the code.  I
think the laser should be off when traveling(?), such as the 10mm
acceleration zone you have implemented?  (Perhaps I need to understand
better).

Thank you, so glad that you are working on this

-- 
Bruce Chadbourne
306 Fern Circle
Lebanon PA 17042

rktech-de, 2022-01-10:
Hi Bruce,
Now we are talking here about my fork in this foreign repository, could you please put your comments as separate issues in my fork-repository? THX

BCChad, 2022-01-10:
Yes, will do that later today.